Pr20100/SOVOL-SV06-Klipper-profile #3: avoid StealthChop/SpreadCycle switching with stealthchop_threshold

fex42, 2023-01-08:
The [Klipper documentation](https://www.klipper3d.org/TMC_Drivers.html#setting-spreadcycle-vs-stealthchop-mode) advises against switching between StealthChop and SpreadCycle while driving (which is exactly what a stealthchop_threshold parameter > 0 does).
 
In fact, my printer makes very unhealthy noises during acceleration runs when the stealthchop_threshold threshold is exceeded (it sounds like something is mechanically broken). According to the documentation, this cannot also cause undefined behavior (errors).

Just because Sovol does this wrong in their stock Marlin FW, it should not be adopted in Klipper. It is better in this case to set "stealthchop_threshold: 999999", which practically prevents switching to SpreadCycle.

Pr20100, 2023-01-09:
Hi, with stealthchop_threshold: 999999 it's also better to turn on interpolation. I don't like this solution but I will make a try with it and another with stealthchop_threshold: 1 with no interpolation, it could be a better option. If you can try it too and make a feedback it will be helpful.

Pr20100, 2023-01-09:
New printing profiles added on pre-release version

closed 2023-01-09 by Pr20100