Pr20100/SOVOL-SV06-Klipper-profile #5: Add M204 gcode macro in macro.cfg to change within a print to different access/deaccel values

MPC561, 2023-01-11:
The M204 code instruction is used to set access/deaccel values in some slicers and not known by Klipper.

There are multiple benefits to have this macro. 
1.) The Pressure Advance uses this M204 instruction to set access to a defined value
2.) Also some Slicers (at least SuperSlicer and Prusaslicer, with Prusaslicer I tested successfully that it works) use the M204 command. The benefit is that in Prusaslicer you can set multiple acceleration values for different feature types, for instance separately for Perimeters, Infill, Bridges and the First Layer. So you can use a small acceleration value for the Perimeters to avoid visible ringing and increase the acceleration value for infill where ringing is not a problem because not visible

I found a appropriate macro in following GitHub repository: [Click me](https://github.com/Desuuuu/klipper-macros/tree/master/macros)

Here the content of the macro:
```
[gcode_macro M204]
rename_existing: M204.1
gcode:
  {% set f = params.F|default(0.5)|float %}

  {% if 'S' in params %}
    {% set s = params.S|float %}
    SET_VELOCITY_LIMIT ACCEL={s} ACCEL_TO_DECEL={ s * f }
  {% else %}
    {% if 'P' in params %}
      {% set p = params.P|float %}
      {% if 'T' in params %}
        {% set t = params.T|float %}
        {% if p < t %}
          SET_VELOCITY_LIMIT ACCEL={p} ACCEL_TO_DECEL={ p * f }
        {% else %}
          SET_VELOCITY_LIMIT ACCEL={t} ACCEL_TO_DECEL={ t * f }
        {% endif %}
      {% else %}
        SET_VELOCITY_LIMIT ACCEL={p} ACCEL_TO_DECEL={ p * f }
      {% endif %}
    {% elif 'T' in params %}
      {% set t = params.T|float %}
      SET_VELOCITY_LIMIT ACCEL={t} ACCEL_TO_DECEL={ t * f }
    {% endif %}
  {% endif %}
```

Pr20100, 2023-01-12:
Hello, thanks for the feedback. Not needed if you set the good g-code flavor setting in SuperSlicer

closed 2023-01-12 by Pr20100

MPC561, 2023-01-12:
Not really. This is true for the Acc values but not for the deacc values. The macro sets the deacc also to a factor depending on the acc value.