Pr20100/SOVOL-SV06-Klipper-profile #7: Issues with Multi Klipper

Twitch0815, 2023-01-16:
I am using multiple instances of Klipper which organizes different printers as  printer_1_data, printer_2_data, etc etc for how many instances you have

With the first version of files printer.cfg and printer profiles were the only paths that had to be adjusted

but with these updates I tried and cannot seem to find all the paths or ways to adjust the paths to get this to work.  

Can you make this friendly for multiple printer installs

think part of the problem is around
 /home/pi/klipper_config/config/ = /home/pi/printer_1_data/config/config/ (edited)

eg, /home/pi/klipper_config/config/options.cfg = /home/pi/printer_1_data/config/config/options.cfg

Pr20100, 2023-01-16:
Hello, it's not possible to make it work in this way. For you it's maybe printer_1, for someone else will be printer_3... I've also already shortened the syntax path for the files, ex : 
NOW [include /config/drivers/TMC_2209.cfg]
BEFORE [/home/pi/printer_data/config/config/drivers/TMC_2209.cfg]
It's not possible for the printing-profiles files.
So if your're using the last update, you will have to try to change only the 3 files path in the "config/printing-profiles/ X-file" (2 lines for each) it should work. Can you make me a feedback?

Twitch0815, 2023-01-16:
Sure I can try,  and give feedback 
Need to figure out what machine Gcode to change my cura as I was using marlin a long time not sure how to edit the gcode for klipper specifically.

Also is there a way to set z offset with the screen at the printer?

Pr20100, 2023-01-16:
I have 2 things to do and this is it!
Will update the display cfg to add all the macros in a menu
Will make a file to give the start/end print gcode for the slicers (Prusa/Superslicer/Cura) cause there's some differences for each

Twitch0815, 2023-01-17:
question,
with the newest release there is a firmware.bin file.  If we have already updated or kipper firmware on the printer do we need to reflash this, or is that unneeded? 
https://github.com/Pr20100/SOVOL-SV06-Klipper-profile/releases/tag/v120123

Twitch0815, 2023-01-17:
Thus far the new file release setup is a lot easier to understand and the pathing.  Updating files didnt break connection to the printer.  I am also able to easily update the path information.  
Only thing is now I have a printer.cfg in both /home/pi/printer_2_data/ and /home/pi/printer_2_data/config  
Not sure which one is authorotative

Twitch0815, 2023-01-17:
Alright I am still getting an error message though I have edited  all files in printer_profiles and printer.cfg

Error I am getting is 
GCode path received from Klipper does not match expected location.

Received: '/home/pie/printer_2_data/gcodes'
Expected: '/home/pi/printer_2_data/gcodes'

Modify the [virtual_sdcard] section Klipper's configuration to correct this error.

[virtual_sdcard]
path: /home/pi/printer_2_data/gcodes

closed 2023-01-17 by Pr20100