Pr20100/SOVOL-SV06-Klipper-profile #10: Request: Can you please add the Ellis TEST_SPEED macro?

bassamanator, 2023-01-29:
You can find it here: https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/debf421280af706316de2ad014333119b6db6607/macros/TEST_SPEED.cfg

I have it working on my Voron 0.1, however, the one time I tried it on the SV06, it didn't go well: 
- After the macro did the test run, it decided to home Z (I suspect) but the nozzle crashed into the bed instead.

Thanks!

Pr20100, 2023-01-30:
Hi, as it's not a usefull fonction for most user it will not be implemented in a next release. If you make it working properly, you can share the config file with me, I'll try it and add it if it's working to an "optional macros" page

bassamanator, 2023-02-03:
Well, considering the excellent way in which you've split up this configuration, with all the different profiles from basic to aggressive, I would say that this macro belongs in this repo. In any case, I will try to get it working and create a PR. 

Closing for now.

closed 2023-02-03 by bassamanator